vrruiz/visualino #6: Debian package errors

vrruiz, 2015-04-06:
dpkg-buildpackage: source-only upload: Debian-native package
Now running lintian...
W: visualino source: no-section-field-for-source
E: visualino source: source-is-missing build-src-Desktop-Debug/visualino
E: visualino source: source-is-missing roboblocks/html/javascript/blockly-bq/arduino_compressed.js
E: visualino source: source-is-missing roboblocks/html/javascript/blockly-bq/blockly_compressed.js
E: visualino source: source-is-missing roboblocks/html/javascript/blockly-bq/blocks_compressed.js
E: visualino source: source-is-missing roboblocks/html/javascript/blockly-bq/javascript_compressed.js
E: visualino source: source-is-missing roboblocks/html/javascript/jquery/dist/jquery.min.js
E: visualino source: build-depends-on-metapackage build-depends: qt5-default
W: visualino source: missing-license-paragraph-in-dep5-copyright apache (paragraph at line 9)

vrruiz, 2015-04-06:
Without qt5-default, it will complain about qt version not defined and won't build.

ctemescw, 2015-07-01:
Hi,

There is another problem with the Debian package, as it also includes parts of the directory where it has been built in the final package.

Example: The package for the i386 architecture currently available on the web has the "tmp" directory inside.

ctemescw@jbuild32:/home/devel/sources/visualino/CONTENTS$ tree tmp
tmp
└── buildd
    └── visualino-0.5
        └── src
            ├── config.ini
            └── html
                ├── blockly.extensions.js
                ├── constants.js
                ├── COPYING.LESER.md
                ├── COPYING.md
                ├── img
                │   └── blocks
                │       ├── bqmod01.png
                │       ├── bqmod02.png
                │       ├── bqmod03.png
                │       ├── bqmod04.png
                │       ├── bqmod05.png
                │       ├── bqmod06.png
                │       ├── bqmod07.png
                │       ├── bqmod08.png
                │       ├── bqmod09.png
                │       ├── bqmod10.png
                │       ├── bqmod11.png
                │       ├── bqservo01.png
                │       ├── bqservo03.png
                │       ├── lcd.png
                │       ├── zum01.png
                │       ├── zum02.png
                │       ├── zum03.png
                │       ├── zum04.png
                │       ├── zum05.png
                │       ├── zum06.png
                │       └── zum07.png
                ├── index.html
                ├── javascript
                │   ├── blockly-bq
                │   │   ├── arduino_compressed.js
                │   │   ├── blockly_compressed.js
                │   │   ├── blocks_compressed.js
                │   │   ├── COPYING
                │   │   ├── javascript_compressed.js
                │   │   └── README.md
                │   ├── jquery
                │   │   ├── dist
                │   │   │   ├── jquery.min.js
                │   │   │   └── jquery.min.map
                │   │   └── MIT-LICENSE.txt
                │   ├── requirejs
                │   │   └── require.js
                │   └── underscore
                │       ├── LICENSE
                │       ├── README.md
                │       └── underscore.js
                ├── lang.js
                ├── media
                │   ├── 1x1.gif
                │   ├── anon.jpeg
                │   ├── click.mp3
                │   ├── click.ogg
                │   ├── click.wav
                │   ├── delete.mp3
                │   ├── delete.ogg
                │   ├── delete.wav
                │   ├── handclosed.cur
                │   ├── handopen.cur
                │   ├── progress.gif
                │   ├── quote0.png
                │   ├── quote1.png
                │   └── sprites.png
                ├── profiles.js
                ├── README.md
                ├── roboblocks.js
                ├── roboblocks.min.js
                └── visualino.css

13 directories, 60 files

vrruiz, 2015-07-01:
Hi Carlos. I'll take a look.

vrruiz, 2015-07-03:
For some reason, pbuilder-dist is including the /tmp directory, I
don't understand why.

ctemescw, 2015-07-03:
I have very little experience in software packaging for Debian/Ubuntu distros, but I've found a temporary solution for this problem. In the "debian/rules" file I've set the following lines:

>override_dh_fixperms:
>  rm -rf $(CURDIR)/debian/visualino/home
>  dh_fixperms

Lt-Henry, 2020-03-25:
When deb package is populated, debhelper tools automatically perform a qmake, make and make install. QMake file src.pro provides a INSTALL= clause, and there is also a INSTALL file at debian directory, which I guess it contains what @vrruiz really wanted to package. This issue can be solved by either using qmake install mechanism or debian provided one.

Besides that, some files are deployed in wrong places. config.ini must be in /etc, not /usr/share, and the icon is better at /usr/share/icons/hicolor/

I will try to submit a pull request

commit b4f9322f (2020-05-02): Merge pull request #52 from edupals/fix_6

fix #6
closed 2020-05-02 by vrruiz