vrruiz/visualino #8: Great !!!!

SebCanet, 2015-04-06:
Hi,
I created a totally BlocklyDuino, from new blockly core with internationalization (spanish version is made by José Manuel Ruiz, thanks to him), and I tried my version in the html directory : it works !!!!
https://github.com/technologiescollege/BlocklyDuino_tests
You also created a Bitbloq copy, I'll have to look to take it inside my version, that's great !

I hope you will find my version interesting and maybe we could work together on the Blockly version.

Thanks a lot for your work, that's great, exactly what I couldn't create !

vrruiz, 2015-04-10:
Hi Seb! Thanks :) If you take a look to the old commits, you'll see that originally the project started with BlocklyDuino, so we share a common ancestor! However, at the time it lacked internationalization support among other things. I've taken a look to your BlocklyDuino site, and looks great.

Right now, my plan is to stick to Roboblocks for the main version. bitbloq/roboblocks is already translated to many languages and bq is actively maintaining the platform -although they have started working on a completely new javascript engine. However, I'm open to discuss creating an alternate Visualino distribution with BlocklyDuino.

SebCanet, 2015-04-11:
Bitbloq works on something different than Blockly engine ??
I started from scratch with the new blockly engine, added blocklyduino's grove block, added translation and new blocks. I think the most interesting in your project i the ability to embed any kind of web project in a software project.

How can I add my work (blocks) in your version ?
Thanks.

SebCanet, 2016-12-13:
I finished my version a while: @Blockly@rduino [http://blocklyarduino.technologiescollege.fr/](http://blocklyarduino.technologiescollege.fr/)
I tried to use visualino to embed my work in an executable but I have a few problem., I tried to do mine with my old copy of Borland C++: [https://github.com/technologiescollege/Blockly-rduino_AIO](https://github.com/technologiescollege/Blockly-rduino_AIO), but I prefer what you did with Qt.

I don't know if if you continue to stick to Roboblocks, since they do something completely different from Blockly ! My project is very used in France, I know from Spain, Italy, China, US and whole Quebec. And they all ask for something like Visualino.