vrruiz/visualino #9: Add support for zum board (Arduino BT)

Obijuan, 2015-04-15:
Currently only arduino uno, mega and nano are supported.  The zum board is compatible with Arudino BT. It allows to download programs through bluetooth. It is great for teaching robot programing to kids.

It would be great if the zum board is added to visualino

Great tool! Thanks! :-)

Obijuan, 2015-04-15:
More information.

The only difference between arduino uno and zum board is that the baudrate should be changed to 19200 by means of the param:   -b19200  when calling avrdude

This is what is executed when uploading to the zum:

avrdude -C/home/obijuan/arduino-1.6.3/hardware/tools/avr/etc/avrdude.conf -q -q -patmega328p -carduino -P/dev/ttyUSB0 -b19200 -D -Uflash:w:/tmp/build2154442348667408074.tmp/Blink.cpp.hex:i

This is what is executed when uploading to arduino uno:

avrdude -C/home/obijuan/arduino-1.6.3/hardware/tools/avr/etc/avrdude.conf -q -q -patmega328p -carduino -P/dev/ttyUSB0 -b115200 -D -Uflash:w:/tmp/build2154442348667408074.tmp/Blink.cpp.hex:i

Obijuan, 2015-04-15:
I've read the code and now I undestand it a little better :-)  For uploading code to arduino boards Visualino is calling arduino ide.  I've successfully upload code to zum board with this command in the console:

arduino --upload --board arduino:avr:bt --port /dev/ttyUSB0 /tmp/visualino/visualino.ino

So, for implementing the zum board only is needed to identify it as:  "arduino:avr:bt"  :-)

Obijuan, 2015-04-15:
Success!!!  :-)   I have emitted a pull request with the fix

vrruiz, 2015-04-15:
Thanks a lot! This is the first contribution to the IDE itself, wee! :))

I've merged the pull request, but I think support is still needed to set the baud rate on the command line. Can you confirm?

vrruiz, 2015-04-15:
I see that boards.txt define bt.upload.speed=19200 by default, so no need.

Obijuan, 2015-04-15:
I will close this issue. Thanks a lot!  :-)

closed 2015-12-04 by vrruiz