vrruiz/visualino #12: LCD block

fjtrsq, 2015-07-28:
Hello. the lcd block don't work properly, i don't know if the library is the correct one. Any idea?

vrruiz, 2015-07-28:
Can you give more details? Which LCD are you using? Are you getting any
error?

fjtrsq, 2015-07-29:
I'm  using an LCD bq bought in and gives me a compile error

fjtrsq, 2015-07-29:
A partir de aqui voy a seguir en castellano,si es posible.
El LCD es el que vende bq en su tienda, ahora no se el modelo exacto y me sale un error de compilacion.
Por otro lado felicitarte por tu programa, justo el que he estado buscando, pues aunque me gustaba el bitbloq de bq no me terminaba de funcionar bien a la hora de cargar en la placa, cosa que si me hace Visualino. 
De las pocas veces que me funcionó Bitbloq pude programar la LCD, por eso entiendo que funciona.
Ademas he visto que el codigo arduino generado no se incluyen las librerias en las sentencias #include., o quiza estan ocultas...

vrruiz, 2015-07-29:
Muchas gracias. Me alegra mucho que Visualino te sea útil :))

Acabo de comprobar, y efectivamente, hay un error en la inicialización del
objeto LiquidCrystal.

#include <Wire.h>

#include <LiquidCrystal.h>

/***   Global variables   ***/

LiquidCrystal lcd(0);

visualino.ino:5:20: error: no matching function for call to
'LiquidCrystal::LiquidCrystal(int)'

Debería ser algo así como:

LiquidCrystal lcd(12, 11, 10, 5, 4, 3, 2);

fjtrsq, 2015-07-30:
Gracias por contestar. Y ¿cual sería la solución para que funcionase?
Porque no se puede cambiar el código desde dentro de Visualino y mi
intención es utilizar el programa con mi hijo de 10 años por lo que hacerlo
directamente desde arduino va a ser muy complicado....
Por cierto en la sección de código no aparecen las librerías en las
sentencias #include, no se si esto es lo normal
Javier.

vrruiz, 2015-07-30:
La solución es arreglar la generación automática de código. Espero
mirármelo lo antes posible.

Estoy casi seguro de que lo de la sección de código es un problema con los
caracteres de mayor y menor < >, que se están interpretando como etiquetas
HTML. También lo miraré :)

vrruiz, 2015-07-30:
Sobre el problema con el LCD, está identificado y reportado en https://github.com/bq/roboblocks/issues/15 

El otro problema de mostrar las librerías de los #include ya está solventado.

fjtrsq, 2015-07-31:
Perdona mi insistencia e ignorancia, pero ¿que se supone que tengo que hacer para solventar el error?
He visto que en la pagina de Visualino.net has incluido un pequeño archivo visualino-exe-zum-test. No se si con instalarlo basta o hay que hacer algo mas. Hasta esta tarde no podré probarlo, pero por ir adelantando... je,je,je.
Muchas gracias.
Javier.

vrruiz, 2015-07-31:
No tienes que hacer nada, Javier, ya lo estoy intentando resolver con los
colegas de bq.

Ese fichero zip no está relacionado con tu problema. Contiene un ejecutable
que habilita la Zum en el menú desplegable de placas.

alviboi, 2017-07-20:
Hola,

Yo lo he solucionado en el archivo /usr/share/visualino/html/roboblocks.js:

Quitas la siguiente línea 5282 :

__p += '#include <Wire.h>\n #include <LiquidCrystal.h>';

Y la substituyes por:

 __p += '#include <LiquidCrystal.h>';

Y ya está.

Me acabo de instalar Visualino y me he encontrado con este mismo problema.

Saludos