vrruiz/visualino #19: Windows: Save and open same program duplicates the blocks

vrruiz, 2015-12-17:
Test case.
- Create a sketch in Visualino 0.6.
- Save the sketch.
- Open the same sketch.

Expected result.
- Sketch is correctly opened.

Actual result.
- All blocks are duplicated.

vrruiz, 2015-12-17:
For some reason, JST is escaping nested if/else blocks. Generating the code using direct JavaScript avoids the problem:

```javascript
    code += 'if (' + argument + ') {\n' + branch + '}';
    /*
    code += JST['controls_if']({
        'argument': argument,
        'branch' : branch
    });
    */
```

Lt-Henry, 2020-04-02:
Hey! A teacher just mail me with a visualino project where there are two empty main blocks (setup-loop) and a third one where the teacher works on. Those empty blocks cannot be deleted, and code generation is messed up. Loop code is properly generated but setup part is not, but setup code is generated from one of those empty blocks.

I have tried to fix it with a text editor removing those extra blocks with no luck, visualino does not load anything. I even tried switch deletable flag to true with same result: nothing loads.
[duplicate.txt](https://github.com/vrruiz/visualino/files/4423284/duplicate.txt)

Lt-Henry, 2020-04-08:
Ok, I am not even close to figure it out but at least, I got the project working editing xml with a text editor. Perhaps... xml parser does not allow \n characters?

vrruiz, 2020-04-09:
I think in old versions there was a way to add multiple main blocks
(an error), but not to remove them. I think I removed that option. And
the solution is to remove the duplication manually... I know, that
sucks.