vrruiz/visualino #21: Translations?

m1guelpiedrafita, 2016-04-11:
Does this project support translations?
If yes, What languages do you have
If not, maybe I can help translating it to Spanish

vrruiz, 2016-04-11:
Hola Miguel. Sí, este proyecto soporta traducciones y, por desgracia, ya
está traducido al español. De todas formas, si quieres revisarlas, los
ficheros de traducción andan en el directorio /ts/ (hace falta el programa
Qt Translator para editarlos).

tecnoloxia, 2016-11-07:
Como tengo que hacer para aportar una traducción a galego?
Gracias

vrruiz, 2016-11-07:
Puedes modificar los ficheros en portugués (o en español). Hay dos
cosas a traducir: los bloques y el programa.

- Para traducir los bloques puedes mirar en este directorio
https://github.com/vrruiz/roboblocks/tree/master/lang Sólo tienes que
copiar uno de esos ficheros como gl-es.js.
- Luego, para traducir el resto del interfaz puedes usar Qt
Linguistic. Los ficheros de traducciones del interfaz andan por aquí:
https://github.com/vrruiz/visualino/tree/master/ts

Si tienes más dudas, puedes resolverlas por aquí o en los foros.

tecnoloxia, 2016-11-07:
Ok, ya tengo los ficheros. No me permite subilos aquí. Están en [google drive](https://drive.google.com/file/d/0B6CqFl4gn5RCS2xPUURGemxrb3c/view?usp=sharing).

m1guelpiedrafita, 2016-11-07:
@tecnoloxia Te los subía Github.
Enlaces: https://github.com/vrruiz/roboblocks/pull/4
https://github.com/vrruiz/visualino/pull/22

vrruiz, 2016-11-07:
Gracias por la traducción y por enviarlo a github. ¡Están ya en los repos!
En cuanto pueda genero una nueva versión con Visualino en gallego.

m1guelpiedrafita, 2016-11-07:
Closing issue. You can reopen it if you need

closed 2016-11-07 by m1guelpiedrafita

tecnoloxia, 2016-11-08:
Muchas gracias. Estoy preparando unas prácticas con visualino y me gustaría contar con el software en galego para la documentación.