vrruiz/visualino #24: error visualino 0.7 en ubuntu 16.10 no sube el codigo

aguscout, 2016-11-12:
Buenas noches, tengo un problema con el visualino 0.7 en Ubuntu 16.10, se instala correctamente y abre correctamente , cuando voy a probar un simple blink por puerto 12 , no logra llevar el código a la placa arduino... tampoco genera errores.
Como prueba complementaria el mismo código generado lo probé en el ide de arduino version 1.6.12 y va correctamente. 
También confirme que el puerto sea el mismo que el del ide arduino y que no estén ambas aplicaciones abiertas...
Respecto a vuestro software pues probé muchos ide y la verdad es increíble que tengas una solución tan bonita y potente.. espero que tengas nuevas versiones programadas, definitivamente le pasa el trapo al Snap4arduino... muy bien.
quiero traspasar varios proyectos que ya tengo armados en s4a y en el ide de arduino pero necesito poder dejarlo funcionando en el ubuntu,,,

vrruiz, 2016-12-13:
¡Muchas gracias! Y perdona por el retraso en contestar, no se por qué, pero no había visto el mensaje. ¡Mil perdones! He probado con Ubuntu 16.10 y 1.6.13 y aquí funciona correctamente. ¿La consola no da ningún tipo de error? Al menos debería mostrar los mensajes de compilación.

aguscout, 2016-12-13:
Gracias por responder Victor, pues mira viendo tu respuesta me di cuenta 
que que es como dices, verifica bien y cuando presiono para subir el 
codigo aparece el siguiente mensaje que segun entiendo es porque no 
estoy como root , pero mi usuario si es mira como tengo configurado el 
acceso directo... cada vez que abro me pide pass de root como 
corresponde, es raro lo que pasa, si ves algun detalle a corregir te 
pediria que me lo indiques para solucionarlo..

Te adjunto en texto el texto de respuesta al intentar subir el codigo y 
la pantalla donde te muestro las propiedades del acceso directo.. Muchas 
gracias y todo bien con las demoras .. Atte Agustin

Compilando...

Picked up JAVA_TOOL_OPTIONS:

Cargando configuración...

(java:9330): IBUS-WARNING **: The owner of /home/agu/.config/ibus/bus is 
not root!

java.lang.NullPointerException

at processing.app.BaseNoGui.selectSerialPort(BaseNoGui.java:1088)

at 
processing.app.helpers.CommandlineParser.parseArgumentsPhase1(CommandlineParser.java:143)

at processing.app.Base.<init>(Base.java:273)

at processing.app.Base.guardedMain(Base.java:219)

at processing.app.Base.main(Base.java:132)

Finalizó.

-- Agustín Jaume Móvil : 675218393 agustin@practicard.es Soluciones 
tecnológicas https://es.linkedin.com/in/agustin-jaume-617b3733

vrruiz, 2016-12-14:
Lo del permiso es sólo un aviso, esa parte no es el problema. El problema está en esta otra línea.

processing.app.helpers.CommandlineParser.parseArgumentsPhase1(CommandlineParser.java:143)

Por lo que comentaban en https://github.com/arduino/Arduino/issues/5413 era un problema con la versión 1.6.12 del IDE de Arduino. ¿Puedes probar con 1.6.13?

aguscout, 2016-12-14:
> Por lo que comentaban en arduino/Arduino#5413

ok , pruebo y te digo.. muchas gracias... te tengo al tanto..

atte
Agustin

aguscout, 2016-12-14:
Perfecto ,, no me aguante a probar... ya actualice y sube bien el codigo 
, al menos el blink ...mañana pruebo un circuito con mas codigo muchas 
gracias...Victor..

mira el codigo de respuesta..pero al menos el codigo si pasa... bien...

Saludos Agustin

Compilando...

Picked up JAVA_TOOL_OPTIONS:

Cargando configuración...

(java:14642): IBUS-WARNING **: The owner of /home/agu/.config/ibus/bus 
is not root!

Inicializando paquetes...

Preparando tarjetas...

Verificando y cargando...

El Sketch usa 928 bytes (2%) del espacio de almacenamiento de programa. 
El máximo es 32.256 bytes.

Las variables Globales usan 9 bytes (0%) de la memoria dinámica, dejando 
2.039 bytes para las variables locales. El máximo es 2.048 bytes.

Finalizó.

-- Agustín Jaume Móvil : 675218393 agustin@practicard.es Soluciones 
tecnológicas https://es.linkedin.com/in/agustin-jaume-617b3733

Lt-Henry, 2020-03-16:
Yo he detectado otra condición en la que aparentemente se verifica y se sube, pero luego no se sube lo que toca. Por defecto se genera un archivo temporal en /tmp/visualino/visualino.ino Si otro usuario en la misma máquina ha usado visualino anteriormente, este archivo temporal no se puede sobreescribir. Propongo que aunque la ruta temporal sea configurable, se genere un archivo unico por usuario, ej: /tmp/visualino/fulanito.ino