vrruiz/visualino #28: Porting from Qt WebKit to Qt WebEngine

SebCanet, 2016-12-13:
Hi,
after trying to reinstall QT (v5.7), I also decided to try to understand your Qt code but there's a version problem:

> Qt WebEngine supercedes the Qt WebKit module, which is based on the WebKit project, but has not been actively synchronized with the upstream WebKit code since Qt 5.2 and has been deprecated in Qt 5.5. For tips on how to change a Qt WebKit widgets application to use Qt WebEngine widgets, see Porting from Qt WebKit to Qt WebEngine. For new applications, we recommend using Qt Quick and the WebEngineView QML type.

That's why I cannot compile your visualino...
I wanted to help you and add some functionnality, like Arduino auto-detect, but I can't.

Hope you can solve it soon.
Thanks !

vrruiz, 2016-12-13:
That would be great, Seb! I'll take a look as soon as possible to migrate to Qt 5.7.

vrruiz, 2016-12-13:
Weird. I'm developing on Ubuntu 16.10 using Qt 5.6 and there are no problems during the build regarding the QWebView.

SebCanet, 2017-01-17:
If you want to see how students crated their autodetetc of Arduino, it's all inside : [https://github.com/technologiescollege/openorganigram/tree/master/src/Interface/LibSerialPort](https://github.com/technologiescollege/openorganigram/tree/master/src/Interface/LibSerialPort)

I'll try again with Qt...

vrruiz, 2017-01-17:
Weeks ago, I created a branch with the ongoing work to port Qt WebKit to Qt WebEngine. The new API is not as good and complete, and found some problems along the way.

vrruiz, 2017-01-17:
Where is the code to autodetect Arduino in those files? Visualino detects the active serial ports, but not which of them are actually Arduino.

SebCanet, 2017-01-17:
In the different qextserialenumerator.
It uses vendorID to verify if it's Arduino board.