vrruiz/visualino #29: Visualino for Blockly@rduino ?

SebCanet, 2016-12-13:
I finished my version a while: @Blockly@rduino http://blocklyarduino.technologiescollege.fr/
I tried to use visualino to embed my work in an executable but I have a few problem., I tried to do mine with my old copy of Borland C++: https://github.com/technologiescollege/Blockly-rduino_AIO, but I prefer what you did with Qt.

I don't know if if you continue to stick to Roboblocks, since they do something completely different from Blockly ! My project is very used in France, I know from Spain, Italy, China, US and whole Quebec. And they all ask for something like Visualino.

Hope we could work together.

vrruiz, 2016-12-13:
Hi Seb! Indeed, I've been planning a migration to stock Blockly to take advantage of some of the recent developments (like the zoom feature), but haven't found the time. So let's talk! I'll take a close look to your project :)

SebCanet, 2016-12-14:
Thanks, hope you''l like. We work a lot on it to make it easy for newbie.
It lacks spanish translation...

SebCanet, 2017-01-17:
Did you have a look ? I don't know exactly how QTWebBrowser works but it seems to have problems with Ajax...

SebCanet, 2017-02-25:
The embedded webbrowser is quite strange... everything is not working.
But if I try launching throught nwjs there is no problem.
Still working on it...