vrruiz/visualino #32: Visualino para arduino nano 168

bercebus, 2017-04-10:
¡Hola!

Me preguntaba si sería posible que Visualino pudiese funcionar para arduinos nano con el atmel 168, ya que tenemos un buen puñado de ellos pero no los podemos usar con esta fantástica herramienta.

Un saludo y muchas gracias

vrruiz, 2017-04-10:
Hola. Sí, desde luego que es posible. Lo miro y hago un parche :)

bercebus, 2017-04-10:
¡Genial, muchas gracias!

TitoVermislab, 2019-03-06:
Muy buenas, ese parche es para la version de windows? donde puedo encontrarlo? es que tengo el mismo problema. Veo que en la version de linux tienes para en 328 oldbootloader, pero en la version de windows no sale esa opcion. Muchas gracias

vrruiz, 2019-03-06:
El parche no está todavía en ninguna versión publicada, pero debe
servir también para Windows.

TitoVermislab, 2019-03-06:
Entonces no puedo usar el Nano 168 ni 328 oldbootloder en el visualino para windows?