vrruiz/visualino #36: Impossible to select filename when saving

gmenegoz, 2017-05-30:
I installed Visualino 0.7 for MacOS and compiled from source for ArchLinux: in both cases when I try to save I cannot choose the filename.  If I want to overwrite on an existing .bly file everything works well but when I need to specify a new name I can't. The only thing it allows me to do is to select a folder (let's say "FolderName"), but if I press Save, instead of creating a new file in that folder, it produces a file FolderName.bly in the home directory.
Is it a Qt5 problem with the dialog box?

flauta, 2017-11-16:
Found the same problem. I can't specify the file name for saving or exporting, and the menu entry "Save as..." is always greyed. I think maybe it's a problem with QT interface.
[image]
[image]

Manjaro Linux x64

Lt-Henry, 2020-03-16:
Hi, I've already fix it! May I submit a pull-request?