vrruiz/visualino #39: Blockly part isn't loading in application after deployment

disheet, 2017-08-31:
i downloaded visualino source code, after executing the visualino i  am not getting Blockly window [ left side block].

please help me to resolve this issue

i am using
- Qt Creator version  3.2.1(Opensource) based on Qt 5.3.2( MSVC2010, 32 bit )
- Mingw version 4.8.2 32 Bit
- Windows Platform

vrruiz, 2017-08-31:
Make sure you have a html/ directory in the build folder containing
Blockly. If not, you can just copy it from the source code.

disheet, 2017-08-31:
When I am running that file (html) in browser blocky is working fine but
while I am debugging in QT creator blocky is not coming.

Thanks & Regards,
Disheet Pithadia

vrruiz, 2017-08-31:
Right. That's because it's not available in the search path. So check the
build directory.
-- 
--
Víctor R. Ruiz
http://rvr.linotipo.es/

disheet, 2017-09-01:
It Works.. 

Thanks & Regards,
Disheet Pithadia

disheet, 2017-09-01:
hello, 
Thanks for your help, Can you please tell me where can i get the xml file or main files for the Blockly which can be editable.
because i want only few blocks in the left side bar, and i want to remove the other blocks in visualino,

please guide me how to proceed.

Thanks & Regards,
Disheet Pithadia

vrruiz, 2017-09-01:
You have that in a separate project: https://github.com/vrruiz/roboblocks

There is a guide about how to create new blocks and compile them here:
https://github.com/Popitas/roboblocks/blob/master/CREARBLOQUES.md
(it's currently in Spanish, use a web translator).