vrruiz/visualino #41: Troubles compiling visualino in Fedora 26

olea, 2018-01-10:
Hi:

Here using visualino from master branch and trying to compile  visualino. We indeed compile it but the make install step doesn't work. I'm not used to the qt framework practices so maybe I'm missing something obvious. All binary dependencies seem to be correct as the binary is generated and runs.

Here is the log of the building process: https://asciinema.org/a/uzsKy2dRgFRf8uZvNsW2ma0ig

Any advise, please?

vrruiz, 2018-01-11:
Try this: sudo mkdir /usr/share/visualino/ && cp -r roboblocks/html/ /usr/share/visualino/

I just took a look and it works.

olea, 2018-01-11:
> Try this: sudo mkdir /usr/share/visualino/ && cp -r roboblocks/html/ /usr/share/visualino/

Well, I'm sure you are fully aware a local software install outside /usr/local/, /opt/, /srv or $HOME is not a systems good practice, neither help to build, for example, an RPM package.

:-)

vrruiz, 2018-01-11:
The question is, does it work or not? Do you now see the blocks? The code
has Linux packaging in mind, and that's why it searches in
/usr/share/visualino/ by default.

itubal, 2018-01-11:
Thankyou. 
The progam runs nicely. And also runs good starting it from QtCreator.

olea, 2018-01-12:
@vrruiz 

> The question is, does it work or not? Do you now see the blocks? 

@itubal confirmed. I can't check because I all my systems are production and can't afford to mesh the filesystem.

> The code has Linux packaging in mind, and that's why it searches in /usr/share/visualino/ by default.

For RPM packaging this is unsuitable. RPM building doesn't use chroot enviroments for %install step and file installations paths should be constructed adding specific macros as in:

    make install DESTDIR=%{buildroot}

both qmake and make are expected to work this way but I can't figure out why can't work in our case:

```
$   make install DESTDIR=/tmp/farfollas
cd ts/ && ( test -e Makefile || /usr/bin/qmake-qt5 -o Makefile /tmp/visualino/ts/ts.pro DESTDIR=/home/olea/visualino-app ) && make -f Makefile install
make[1]: Entering directory '/tmp/visualino/ts'
make[1]: Nothing to be done for 'install'.
make[1]: Leaving directory '/tmp/visualino/ts'
cd src/ && ( test -e Makefile || /usr/bin/qmake-qt5 -o Makefile /tmp/visualino/src/src.pro DESTDIR=/home/olea/visualino-app ) && make -f Makefile install
make[1]: Entering directory '/tmp/visualino/src'
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/config.ini /tmp/visualino/src/config.ini
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/roboblocks/examples/01-variables.bly /tmp/visualino/src/examples/01-variables.bly
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/roboblocks/examples/02-variables-text.bly /tmp/visualino/src/examples/02-variables-text.bly
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/roboblocks/examples/03-variables-serial.bly /tmp/visualino/src/examples/03-variables-serial.bly
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/roboblocks/examples/04-led-blink.bly /tmp/visualino/src/examples/04-led-blink.bly
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/roboblocks/examples/05-led-blink-3.bly /tmp/visualino/src/examples/05-led-blink-3.bly
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/roboblocks/examples/06-led-blink-for.bly /tmp/visualino/src/examples/06-led-blink-for.bly
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/roboblocks/examples/07-led-knight-rider.bly /tmp/visualino/src/examples/07-led-knight-rider.bly
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/roboblocks/examples/08-led-fade.bly /tmp/visualino/src/examples/08-led-fade.bly
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/roboblocks/examples/09-serial-counter.bly /tmp/visualino/src/examples/09-serial-counter.bly
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/roboblocks/examples/10-serial-led-switch.bly /tmp/visualino/src/examples/10-serial-led-switch.bly
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/roboblocks/html/blockly.extensions.js /tmp/visualino/src/html/blockly.extensions.js
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/roboblocks/html/constants.js /tmp/visualino/src/html/constants.js
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/roboblocks/html/COPYING.LESER.md /tmp/visualino/src/html/COPYING.LESER.md
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/roboblocks/html/COPYING.md /tmp/visualino/src/html/COPYING.md
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/roboblocks/html/img /tmp/visualino/src/html/img
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/roboblocks/html/index.html /tmp/visualino/src/html/index.html
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/roboblocks/html/javascript /tmp/visualino/src/html/javascript
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/roboblocks/html/lang.js /tmp/visualino/src/html/lang.js
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/roboblocks/html/media /tmp/visualino/src/html/media
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/roboblocks/html/profiles.js /tmp/visualino/src/html/profiles.js
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/roboblocks/html/README.md /tmp/visualino/src/html/README.md
/usr/bin/qmake-qt5 -install qinstall -exe /tmp/visualino/roboblocks/html/roboblocks.js /tmp/visualino/src/html/roboblocks.js
strip /tmp/visualino/src/html/roboblocks.js
strip:/tmp/visualino/src/html/roboblocks.js: File format not recognized
make[1]: [Makefile:638: install_html] Error 1 (ignored)
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/roboblocks/html/visualino.css /tmp/visualino/src/html/visualino.css
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/ts/visualino_ca-es.qm /tmp/visualino/src/ts/visualino_ca-es.qm
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/ts/visualino_es-es.qm /tmp/visualino/src/ts/visualino_es-es.qm
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/ts/visualino_eu-es.qm /tmp/visualino/src/ts/visualino_eu-es.qm
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/ts/visualino_fr-fr.qm /tmp/visualino/src/ts/visualino_fr-fr.qm
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/ts/visualino_gl-es.qm /tmp/visualino/src/ts/visualino_gl-es.qm
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/ts/visualino_it-it.qm /tmp/visualino/src/ts/visualino_it-it.qm
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/ts/visualino_pl-pl.qm /tmp/visualino/src/ts/visualino_pl-pl.qm
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/ts/visualino_pt-pt.qm /tmp/visualino/src/ts/visualino_pt-pt.qm
/usr/bin/qmake-qt5 -install qinstall /tmp/visualino/ts/visualino_ru.qm /tmp/visualino/src/ts/visualino_ru.qm
make[1]: Leaving directory '/tmp/visualino/src'
sh-4.4$ ls -ld /tmp/farfollas
ls: cannot access '/tmp/farfollas': No such file or directory
sh-4.4$ ls -ld DESTDIR=/home/olea/visualino-app
ls: cannot access 'DESTDIR=/home/olea/visualino-app': No such file or directory
``` 

There is not install error messages neither.

Note that /home/olea/visualino-app path was a parameter when qmake was invoked.

Without modification it can't be packaged in RPM. Would need to check if it can work in Flatpak.