vrruiz/visualino #43: "make: *** No target was specified and no makefile was found. Fedora 27 - Plasma

lea8989, 2018-02-19:
Hello,

I made the following steps:
1- sudo dnf install qt-devel qt5-qtserialport-devel qt5-linguist qt5-qtwebkit-devel

2- Then open the ts.pro file and change all "lupdate" by "lupdate-qt5"
and all "lrelease" by "lrelease-qt5". Save changes

3- cd visualino-master /

4- make
and I receive the message:
"make: *** No target was specified and no makefile was found.

(Excuse my English, I do not handle the language)

Hola,

realicé los siguientes pasos:
1- sudo dnf install qt-devel qt5-qtserialport-devel qt5-linguist qt5-qtwebkit-devel

2- Then open the ts.pro archive and change all "lupdate" by "lupdate-qt5"
and all "lrelease" by "lrelease-qt5". Save changes

3- cd visualino-master/ 

4- make
y recibo el mensaje:
"make: *** No se especificó ningún objetivo y no se encontró ningún makefile.  Alto."

(Disculpe mi ingles, no manejo el idioma)

lea8989, 2018-02-20:
I got it to work with qtcreator.

Sorry for the inconvenience, greetings.

closed 2018-02-20 by lea8989

vrruiz, 2018-02-20:
Hola Leandro. Antes del make debes ejecutar qmake. // Before executing make you must execute qmake.