vrruiz/visualino #47: Visualino no se puede instalar en Fedora 28

lea8989, 2018-11-28:
Hola, 
Realicé todos los pasos de la instalación para poder tener Visualino en Fedora 28 pero no es posible.
Al ejecutar "make" me dice: **make: *** No se especificó ningún objetivo y no se encontró ningún makefile.  Alto**
¿Alguna solución?
Gracias 
Saludos.

lea8989, 2018-11-30:
Hola Victor, ¡gracias por tu respuesta!
Los cambios los había realizado, ahora cambié **make** por **qmake-qt5 && make** y compiló. De todos modos sigo sin poder tener Visualino en Fedora. Luego de los pasos anteriores lo que hice fue:

1. make install
2. sudo mkdir /usr/share/visualino/ && sudo cp -r roboblocks/html/ /usr/share/visualino/

Si accedo a /home/_usuario_/Visualino/visualino-master/src y ejecuto el binario **visualino** me abre el programa pero sin los bloques ([Captura](https://drive.google.com/file/d/1YFN6XErAdc0ukZc0r1xQDRBxPwg4l2pa/view?usp=sharing))
y en /usr/share/visualino no hay ningún binario y el archivo index.html me habre "visualino" en el navegador pero sin la posibilidad de seleccionar las placas ([Captura](https://drive.google.com/file/d/1-03v2pWie9n07JrsJMO64R1R-s0OxJw0/view?usp=sharing))

vrruiz, 2018-12-02:
Justo con ese problema nos hemos topado creando el spec. Hay que modificar la línea de DESTDIR en src/src.pro para indicar la ruta de instalación.