vrruiz/visualino #48: No puedo compilar Visualino en Windows

fpalacios1203, 2018-12-13:
Instale Qtcreator, Qt5.11 y Qt5.4.2, Mingw Qtwebengine y demas pero me sale el mismo error "Unknow module in Qt: webenginewidgets". En KdeNeon no tuve problemas pero en Windows 7 no hay caso. Alguien que pueda darme un mano o si pueden compilar el Visualino para windows y actualizarlo en la pagina de descargas. Gracias

vrruiz, 2018-12-13:
¿Es posible que no hayas instalado el módulo Qt WebEngine?
https://forum.qt.io/topic/92334/unknown-module-s-in-qt-webkitwidgets/13

fpalacios1203, 2018-12-13:
Está instalado. Es más, instale una versión anterior de Qt por las dudas y tampoco me funcionó. la verdad ya no se dónde buscar, porque no trabajo mucho bajo Windows y en Linux funcionó al primer intento

fpalacios1203, 2019-01-29:
Es posible que alguien pueda subir el visualino para windows ya compilado? realmente lo he intentado, pero no he podido bajo windows. Se los agradecería, ya que en breve lo necesitaré para mis clases. Saludos

jenboc3, 2019-02-18:
@fpalacios1203  yo pude compilarlo para windows, si aun lo necesitas.

fpalacios1203, 2019-02-21:
@jenboc3 te lo agradecería, ya que no he podido hacerlo. Tengo windows 7 en una maquina virtual, instale todas las librerías y no he podido hacer que funcione. En linux (KDE Neon), funciono a la primera.

hulkco, 2019-02-21:
> @jenboc3 te lo agradecería, ya que no he podido hacerlo. Tengo windows 7 en una maquina virtual, instale todas las librerías y no he podido hacer que funcione. En linux (KDE Neon), funciono a la primera.

No me hagas mucho caso, pero creo que se puede hace un crosscompiler, y generar el ejecutable para windows desde Linux.

vrruiz, 2019-02-21:
Yo compilo Visualino directamente desde Windows, con el SDK de Qt y Qt
Creator. Seguramente es posible hacer compilación cruzada en Linux, pero no
lo he intentado :)

fpalacios1203, 2019-02-23:
Yo he instalado el sdk, el QT creator y las librerias en una maquina virtual windows 7 y no he podido. En cambio en linux (KDE Neon) ha salido al primer intento. La verdad no se donde está el problema, porque sinceramente nunca utilicé QT Creator

jenboc3, 2019-02-24:
También soy nuevo usando el programa QT creator para compilar. Por razones que aun no comprendo algunas librerías pesan mas de 100 megas por lo tanto no puedo subirlo a Github. Te comparto el link de google Drive donde tengo alojado el programa. https://drive.google.com/open?id=1pQMEZx2YPBiBWbRJXBlDOd23oBZYV2CH

fpalacios1203, 2019-02-25:
Gracias @jenboc3 ! ahi estoy bajando el programa. @vrruiz creo que ya podrías cerrar y dar por resuelto. Gracias a todos!

fpalacios1203, 2019-04-26:
> También soy nuevo usando el programa QT creator para compilar. Por razones que aun no comprendo algunas librerías pesan mas de 100 megas por lo tanto no puedo subirlo a Github. Te comparto el link [...]

Hola @jenboc3 . Ahí pude ver porque pesa tanto el Visualino que compilastes. En varias carpetas, agrega una librería de nombre duplicado, terminada en "d" la cual es mucho mas pesada que la que no posee esa letra. Te paso un captura para que veas. Le quite esas librerías y probé en una maquina virtual y va diez puntos. Muchas gracias
[image]