vrruiz/visualino #53: Issue with temporal files

Lt-Henry, 2020-03-26:
On Linux, the temp file approach has some room for improvement. We are using visualino in a classroom where each kid has his own user and it is common that a computer is used several times without a reboot. What happens then? If user A uses visualino and the logs out, when user B logs in and tries to upload to the board, the temp file (/tmp/visualino/visualino.ino) already exists and belongs to another user. This may be related with some already opened issue!

No error is shown, because file already exists, but it is not overwritten, so project from user A is always compiled and uploaded to the board.

I am not sure how to solve this keeping portability always in mind. IMHO, in Linux I would put visualino.ino file in /run/user/1000/

Yeah, that 1000 is the current user ID, so temp file must be computed  at runtime. But my guess is that a similar approach would be nice also at Windows and OsX.

I propose removing temp config path, and compute a user temp path at runtime.

vrruiz, 2020-03-26:
For that case, you are right. And Qt provides classes for temp files,
should be easy to change.
-- 
--
Víctor R. Ruiz
http://rvr.linotipo.es/

Lt-Henry, 2020-03-26:
Yeah, QTemporaryFile would do the job for all platforms. I can submit a patch, but lets get clear:

- Is privacy important? By default, QTemporaryFile isnt.
- would be nice to print current temp path at stdout for debugging purposes?
- Can we get rid of tmp_dir_name and tmp_file_name?

vrruiz, 2020-03-28:
> Is privacy important? By default, QTemporaryFile isnt.

  Mainly as an educational tool, nope.

> would be nice to print current temp path at stdout for debugging purposes?

  Yeah, QDebug is cool :)

> Can we get rid of tmp_dir_name and tmp_file_name?

  Right, we can get rid of those.

Lt-Henry, 2020-04-08:
I came up to this solution:
https://github.com/edupals/visualino/tree/fix_53

- A random uuid is generated at application start
- path "Qdir::temp() / visualino-UUID / visualino/" is created 
- temp path is removed at exit

This approach allows two instances of visualino to work, even from the same user. I first tried with QTemporaryFile but it does not allow a nested directory as we need (at least, I dont know how).

Worked for me at Linux but It seems it should work fine on both win and mac.

What do you think? ready to merge?

vrruiz, 2020-04-09:
Yeah, I like the solution! The other option would be to generate a
visualino-UUDI.ino file, but it probably makes more sense to isolated it in
its own directory in case there are others artifacts.

Lt-Henry, 2020-04-10:
But arduino ide requires a NAME/NAME.ino pattern, with a uuid would be too long. Let me complete the patch and I expect to submit re request in a couple of days!

vrruiz, 2020-04-10:
Oops, right, I forgot that!
-- 
--
Víctor R. Ruiz
http://rvr.linotipo.es/